micro-bitcoin/uBitcoin #3: Unable to add this library to CMake

baymac, 2019-11-16:
I am trying to add uBitcoin library to my CLion Project and get the following error:

```
Scanning dependencies of target signing_only_wallet
[ 33%] Building CXX object CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o
[ 66%] Building CXX object CMakeFiles/signing_only_wallet.dir/main.cpp.o
[100%] Linking CXX executable signing_only_wallet
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `RMD160::begin()':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:41: undefined reference to `ripemd160_Init'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `RMD160::write(unsigned char const*, unsigned long)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:44: undefined reference to `ripemd160_Update'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `RMD160::write(unsigned char)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:49: undefined reference to `ripemd160_Update'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `RMD160::end(unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:53: undefined reference to `ripemd160_Final'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `sha256Hmac(unsigned char const*, unsigned long, unsigned char const*, unsigned long, unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:74: undefined reference to `hmac_sha256'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA256::begin()':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:79: undefined reference to `sha256_Init'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA256::beginHMAC(unsigned char const*, unsigned long)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:82: undefined reference to `hmac_sha256_Init'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA256::write(unsigned char const*, unsigned long)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:85: undefined reference to `sha256_Update'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA256::write(unsigned char)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:90: undefined reference to `sha256_Update'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA256::end(unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:94: undefined reference to `sha256_Final'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA256::endHMAC(unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:98: undefined reference to `hmac_sha256_Final'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `Hash160::end(unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:121: undefined reference to `sha256_Final'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `DoubleSha::end(unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:145: undefined reference to `sha256_Final'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA512::begin()':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:167: undefined reference to `sha512_Init'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA512::beginHMAC(unsigned char const*, unsigned long)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:170: undefined reference to `hmac_sha512_Init'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA512::write(unsigned char const*, unsigned long)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:173: undefined reference to `sha512_Update'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA512::write(unsigned char)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:178: undefined reference to `sha512_Update'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA512::end(unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:182: undefined reference to `sha512_Final'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `SHA512::endHMAC(unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:186: undefined reference to `hmac_sha512_Final'
CMakeFiles/signing_only_wallet.dir/uBitcoin/src/Hash.cpp.o: In function `sha512Hmac(unsigned char const*, unsigned long, unsigned char const*, unsigned long, unsigned char*)':
/home/parichay/CLionProjects/signing-only-wallet/uBitcoin/src/Hash.cpp:191: undefined reference to `hmac_sha512'
collect2: error: ld returned 1 exit status
CMakeFiles/signing_only_wallet.dir/build.make:98: recipe for target 'signing_only_wallet' failed
make[3]: *** [signing_only_wallet] Error 1
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/signing_only_wallet.dir/all' failed
make[2]: *** [CMakeFiles/signing_only_wallet.dir/all] Error 2
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/signing_only_wallet.dir/rule' failed
make[1]: *** [CMakeFiles/signing_only_wallet.dir/rule] Error 2
Makefile:118: recipe for target 'signing_only_wallet' failed
make: *** [signing_only_wallet] Error 2
```

My `CMakeLists.txt` looks like this (uBitcoin in my root of the project directory):

```
cmake_minimum_required(VERSION 3.15)
project(signing_only_wallet)

set(CMAKE_CXX_STANDARD 14)
set(BOOST_ROOT "/usr/include/boost")

find_package(Boost 1.65.1)

if (NOT Boost_FOUND)
    message(FATAL_ERROR "Could not find boost!")
endif ()

include_directories(uBitcoin/src)

add_executable(signing_only_wallet main.cpp uBitcoin/src/Hash.cpp)

```

stepansnigirev, 2019-11-16:
Try including `src/utility` and `src/utility/trezor`. We are using trezor-crypto library for elliptic curves and hash functions.

baymac, 2019-11-16:
Hi @stepansnigirev, I still get the following error after adding the directories to cmake:

```
CMakeFiles/signing_only_wallet.dir/main.cpp.o: In function `main':
/home/parichay/CLionProjects/signing-only-wallet/main.cpp:219: undefined reference to `sha256(unsigned char const*, unsigned long, unsigned char*)'
collect2: error: ld returned 1 exit status
CMakeFiles/signing_only_wallet.dir/build.make:83: recipe for target 'signing_only_wallet' failed
make[3]: *** [signing_only_wallet] Error 1
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/signing_only_wallet.dir/all' failed
make[2]: *** [CMakeFiles/signing_only_wallet.dir/all] Error 2
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/signing_only_wallet.dir/rule' failed
make[1]: *** [CMakeFiles/signing_only_wallet.dir/rule] Error 2
Makefile:118: recipe for target 'signing_only_wallet' failed
make: *** [signing_only_wallet] Error 2
```

Perhaps I am missing something in linking the library. For reference the source code can be found here: https://github.com/baymac/signing-only-wallet

stepansnigirev, 2019-11-16:
Looks like your compiler doesn't compile `.cpp` and `.c` files other than `main.cpp`.
This CMakeLists.txt worked for me:
```
cmake_minimum_required(VERSION 3.15)
project(signing_only_wallet)

set(CMAKE_CXX_STANDARD 14)

include_directories(uBitcoin/src uBitcoin/src/utility uBitcoin/src/utility/trezor)

file(GLOB_RECURSE CPPSOURCES RELATIVE ${CMAKE_SOURCE_DIR} "uBitcoin/src/*.cpp")
file(GLOB_RECURSE CSOURCES RELATIVE ${CMAKE_SOURCE_DIR} "uBitcoin/src/utility/*.c")

add_executable(signing_only_wallet main.cpp ${CPPSOURCES} ${CSOURCES})
```

I also made a few small fixes that I didn't spot earlier (commit https://github.com/micro-bitcoin/uBitcoin/commit/84556d565d0d917f6310f37fed8864a6e1385861), so update plz :)

baymac, 2019-11-17:
Thanks @stepansnigirev, it works with the new cmake configurations.

closed 2019-11-17 by baymac