micro-bitcoin/uBitcoin #5: Which arduino run this?

he3117, 2020-05-01:
On which arduino can run this library? At the main page its said this library is for arduino but its seems the compiled code is not fit int to the ram of arduino and complier complain about array size is too big. Can we use https://github.com/arduino-bitcoin/arduino-bitcoin or that projects is dead?

stepansnigirev, 2020-05-01:
This library is made to work with Arduino IDE, but 8-bit Arduino microcontrollers are not powerful enough.
It will certainly work on ESP32 and other 32-bit microcontrollers, but not on original 8-bit Arduino.

What are you planning to build? Maybe I can recommend hardware and libraries that would fit your needs.

he3117, 2020-05-02:
I'm trying to build a simple hardware wallet to sign BTC transactions. I have few Ideas about the security model to store seed data on SD card that I like to try.

stepansnigirev, 2020-05-03:
So there are many options for 32-bit microcontrollers with Arduino IDE support:

1. ESP32: the cheapest and very powerful, plenty of devboards available, for example [m5stack](https://m5stack.com/). But they have wifi and bluetooth capabilities and corresponding proprietary code running next to your firmware - so you probably don't want to use it.
2. RiscV based devices like [M5StickV](https://m5stack.com/products/stickv) or [Maix series from Seeed](https://www.seeedstudio.com/Sipeed-MAix-BiT-for-RISC-V-AI-IoT-1-p-2873.html) - they also released a [new Maix Cube](https://www.seeedstudio.com/Sipeed-Maix-Cube-p-4553.html) recently that looks very interesting. Super powerful
3. Adafruit M4 series like [this one](https://www.adafruit.com/product/3857) - they also have plenty of extension boards (feather wings) so it's pretty easy to buy everything you need from them and just plug it together
4. STM32 based boards supported by [stm32duino](https://github.com/stm32duino/Arduino_Core_STM32#boards-available), not sure about blue pill and black pill - they might have not enough RAM and Flash. But other boards should work.

If Arduino IDE is not a requirement, you can also look into Micropython - most of the mentioned boards support micropython in some way. We have a micropython build that includes all bitcoin-related stuff here: https://diybitcoinhardware.com/f469-disco/
Personally I find working with MicroPython very enjoyable.

Regarding the libraries, micro-bitcoin should work fine on all of them, you can also look into [libwally](https://github.com/ElementsProject/libwally-core) from Blockstream and [secp256k1](https://github.com/bitcoin-core/secp256k1) from Bitcoin Core. They are battle-tested and actively maintained libraries, and can be made to work with Arduino IDE.

I didn't work on micro-bitcoin for a while but plan to get back to it soon.

Please let me know if something doesn't work or you have further questions :)

he3117, 2020-05-04:
Thank you for your complete explanation. I decided to use 2 rotary encoder for inputs instead of simple buttons so I think 35stack is not suitable for my design. I ordered ESP-WROOM-32 I think this should be able to run uBitcoin codes. Basically I'm thinking about a new security model to allow me to store secure data on unsecure device like SD card. I have a background in c++ programming but I'm new in hardware space. Current hardware wallets are good against online hackers but it seems they can be vulnerable to hardware hacking. I hope to find a simple way to prevent hardware attacks.

closed 2020-05-05 by he3117

stepansnigirev, 2020-05-05:
Sounds interesting! Keep me posted :)
I would also like to add your project to [this list](https://diybitcoinhardware.com/) if you would open-source it.

he3117, 2020-05-06:
Sure I'd like to help. While I'm waiting for my hardware parts to arrive(because of coronavirus it may take a while) I like to know your opinion about my approach I will open a new topic.