micro-bitcoin/uBitcoin #7: Can not run example sketch

rymkapro, 2020-05-15:
Hi!
I've downloaded .zip from GitHub and added to libraries.
Then I just wanted to upload memonic.ino sketch from examples, and got following errors

```
Arduino: 1.8.12 (Mac OS X), Board: "Arduino Due (Programming Port)"

/arduino/libraries/uBitcoin/src/Conversion.cpp:114:21: error: 'size_t fromHex' redeclared as different kind of symbol
 size_t fromHex(std::string encoded, uint8_t * output, size_t outputSize){
                     ^
In file included from /arduino/libraries/uBitcoin/src/Conversion.cpp:1:0:
/arduino/libraries/uBitcoin/src/Conversion.h:83:8: error: previous declaration of 'size_t fromHex(String, uint8_t*, size_t)'
 size_t fromHex(String encoded, uint8_t * output, size_t outputSize);
        ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:114:16: error: 'string' is not a member of 'std'
 size_t fromHex(std::string encoded, uint8_t * output, size_t outputSize){
                ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:114:45: error: expected primary-expression before '*' token
 size_t fromHex(std::string encoded, uint8_t * output, size_t outputSize){
                                             ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:114:47: error: 'output' was not declared in this scope
 size_t fromHex(std::string encoded, uint8_t * output, size_t outputSize){
                                               ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:114:62: error: expected primary-expression before 'outputSize'
 size_t fromHex(std::string encoded, uint8_t * output, size_t outputSize){
                                                              ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:617:24: error: 'size_t fromBase64' redeclared as different kind of symbol
 size_t fromBase64(std::string encoded, uint8_t * output, size_t outputSize){
                        ^
In file included from /arduino/libraries/uBitcoin/src/Conversion.cpp:1:0:
/arduino/libraries/uBitcoin/src/Conversion.h:103:8: error: previous declaration of 'size_t fromBase64(String, uint8_t*, size_t)'
 size_t fromBase64(String encoded, uint8_t * output, size_t outputSize);
        ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:617:19: error: 'string' is not a member of 'std'
 size_t fromBase64(std::string encoded, uint8_t * output, size_t outputSize){
                   ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:617:48: error: expected primary-expression before '*' token
 size_t fromBase64(std::string encoded, uint8_t * output, size_t outputSize){
                                                ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:617:50: error: 'output' was not declared in this scope
 size_t fromBase64(std::string encoded, uint8_t * output, size_t outputSize){
                                                  ^
/arduino/libraries/uBitcoin/src/Conversion.cpp:617:65: error: expected primary-expression before 'outputSize'
 size_t fromBase64(std::string encoded, uint8_t * output, size_t outputSize){
                                                                 ^
exit status 1
Error compiling for board Arduino Due (Programming Port).
```

Thanks for your help!

stepansnigirev, 2020-05-16:
Thanks! I found a few typos in the definitions of the functions, and fixed them.
It should be good now. Could you re-download the zip and try?

rymkapro, 2020-05-16:
Now everything is perfect!
Thanks a lot!!!

closed 2020-05-16 by rymkapro