micro-bitcoin/uBitcoin #8: This library isn't listed on the PlatformIO

phplego, 2020-12-13:
Thank you for great job and this polished lib! It would be nice if we could download this lib as dependency in PlatformIO project. Thanks

stepansnigirev, 2020-12-15:
Good idea :)
I'll try to find out how to get listed in PlatformIO and arduino libraries list.

phplego, 2020-12-15:
As I understand author have to install PlatformIO to be able to run `pio` command. Then add library.json to the root of the project directory (at the same level as src dir). 

And then execute `publish` command:
```
pio package publish [<source directory, tar.gz or zip>]
```

https://docs.platformio.org/en/latest/librarymanager/creating.html#publishing

stepansnigirev, 2021-01-04:
https://platformio.org/lib/show/11650/uBitcoin

closed 2021-01-04 by stepansnigirev