micro-bitcoin/uBitcoin #9: PSBT.parseBase64(...) failed to parse unsigned transaction generated with Electrum

phplego, 2020-12-13:
Current version of Electrum (4.0.7) does not allow to export unsigned transaction in hex format - only base64. Also it has psbt prefix. So it looks like now only psbt transactions can be generated with Electrum. Just a guess.

For example this is a transaction generated with Electrum (testnet):

`cHNidP8BAHQCAAAAASkA3TuiFr9FXleglmFmRM0JkYOuS3DDZX4ptlqYXVz+AAAAAAD9////AuAuAAAAAAAAGXapFJ+aer1gDAyqA5g6d8jD344GLLL6iKw86BIAAAAAABYAFEPiZyhbzPOiCHHX0oL56PPQvYU2CPQcAAABAPUCAAAAAAEBpTlECzz5cPQe0kSl6C1FrcYk3aUayhaASGKRLB6l8e0BAAAAFxYAFKAxhgGP0V1fYc4jnZZDMNUi+gw9/v///wIKGBMAAAAAABYAFNIExxSjrMyz2cX2BbPwbMyGexsNxGYZ2QEAAAAWABTjkyCFwMo1QHLwIgbQuJs5GMgc+wJHMEQCIHZWy+sOhKYP76WSi6HPpJVs7e2RvN6I8IzeQXp3hbS3AiADt/j73CxxZGNmRvJyWbs3H0Qm/Fc38FOO8PckoSCa+QEhAhMSvrpl3DCjTjP2mWewv40nhAxsgSnelq9H95inoh+YB/QcACIGA5NT4+8EpDjIeFZA2l9Yr+3y3yX9M2F7FpJZ7GqWgs6NDNKCdDUAAAAAAAAAAAAAIgIC1z3a8a4vLjVdvwq1RHv2d0v9qTjXZkj0yAFVfel+cvkM0oJ0NQEAAAAAAAAAAA==
`
[image]

phplego, 2020-12-14:
After some investigation I have found exact place where it fails:

[image]

stepansnigirev, 2020-12-15:
Thanks for reporting! I am a bit busy this week, I'll look into this issue over the weekend.

phplego, 2020-12-24:
> Thanks for reporting! I am a bit busy this week, I'll look into this issue over the weekend.
> 
> I think Electrum switched from their own transaction format to PSBT, so I will also remove Electrum-specific code from the repo.

That you Stepan. Looks like your lib currently is the only one library for doing this on microcontrollers. Would be extremely super helpful to get it working with transactions again (since PSBT). Good luck!

phplego, 2020-12-25:
Further investigation shows that tx hash is not equal:

[image]

stepansnigirev, 2021-01-04:
Should be fixed in https://github.com/micro-bitcoin/uBitcoin/commit/52350a3827c5dc46640e003df07ac41613b4ab39
Can you confirm?

phplego, 2021-01-04:
On ESP32 transaction parsed OK
[image]

On ESP8266 it crushes (out of memory I guess)

closed 2021-01-04 by stepansnigirev