micro-bitcoin/uBitcoin #14: hmac_sha256 conflicts with trezor utilities

claytantor, 2021-11-21:
Hey all, 

I seem to be getting a compile error with multiple definitions of `hmac_sha256` being detected during link

```
/home/clay/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/clay/.arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/lib/libwpa_supplicant.a(crypto_mbedtls.c.obj): in function `hmac_sha256':
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:254: multiple definition of `hmac_sha256'; /tmp/arduino_build_809996/libraries/uBitcoin/utility/trezor/hmac.c.o:/home/clay/data/github.com/claytantor/LNURLPoS/LNURLPoS/libraries/uBitcoin/src/utility/trezor/hmac.c:67: first defined here
collect2: error: ld returned 1 exit status
```
If you go to utilities/trezor/hmac.c

There are 2 definitions

** It looks like one is for confidential addresses and the other for basic

```
void hmac_sha256(const uint8_t *key, const uint32_t keylen, const uint8_t *msg, const uint32_t msglen, uint8_t *hmac)
{
    static CONFIDENTIAL HMAC_SHA256_CTX hctx;
    hmac_sha256_Init(&hctx, key, keylen);
    hmac_sha256_Update(&hctx, msg, msglen);
    hmac_sha256_Final(&hctx, hmac);
}

```

stepansnigirev, 2021-11-22:
Looks like a name collision with mbedtls library...
I'll change the names so it doesn't cause any collisions and report here.

stepansnigirev, 2021-11-30:
Can you test with the current master branch? I think I fixed name collisions.

closed 2021-12-23 by stepansnigirev