micro-bitcoin/uBitcoin #17: [BUG] PSBT cannot sign more than one input

motorina0, 2022-07-27:
### Steps
- Sign a PSBT as shown in the sample code [here](https://github.com/micro-bitcoin/uBitcoin/blob/master/examples/psbt/psbt.ino#L41)
- Make sure that the PSBT has more than one input (the PSBT in sample code has just one)

### Problem
- the scrips sig data is added multiple times to the input
- the return input count is wrong (square of the actual value)

### Further comments
Example of such a transaction (built with uBitcoin):
```
02000000020deeb04344ba2acfd7e1b662604393ca7e8723dc7ef99586c83c52eb1002392600000000d5483045022100a5b40fe13e214a714ee7e524a6e42d65623a4b11b4a2b8bb3c277d6ba3f24f89022036781f9250086ebde04b076531b0ef5937bfcf277f440033450e60b393716cab0121024d3392755edaaff393e2812a5e0d4d98224f56f81dfaaf0d75092c664b9fb3f947304402200d631b820d47d194631a3203237d280a48db8c707c0c6e512a64ce6f9ceb697302200daf260a9c7c55310e3d51cf8d41290d54cbf4f6efe4eb46f27c6194226a03990121028c6f7f55efd404081786745b23ef6b67c42a99f9b163aae45b384164b84c4104ffffffff987e00d9491d4b319c4b65d9e7fa3c7b8063444e91e84a0df851bde3aa7d977904000000d5483045022100d812c0bcc71a1b0f458597150200dfac32d4cba72514fa92fbcd34d2fd168b7602205fd1a1bb8bbdeabc3e272e3e49c62c4471507c3beefa72a0791d9dc4aefaa51e0121024d3392755edaaff393e2812a5e0d4d98224f56f81dfaaf0d75092c664b9fb3f947304402203a12c961f93bb0d38d007c0102cf2ff969d4009c35e3dc7e00e77a1ddc45187b02202dec65aad650e1988675c63aef7a0f7534d5601e859ff071dc632ebbab11cb640121028c6f7f55efd404081786745b23ef6b67c42a99f9b163aae45b384164b84c4104ffffffff02313d0000000000001976a9146e7878188d67b6e0cf101bacfef3ad6a02c8dd5588ac4fa30000000000001976a914eee67e75ce9cbf0d85b4f72ade020c27478efad088ac00000000
```

- `sign()` in `uBitcoin/src/PSBT.cpp`

stepansnigirev, 2022-07-27:
Thank you! I'll look into this issue over the weekend.

stepansnigirev, 2022-07-30:
I was not able to reproduce it. I updated the examples to have 2-input transactions, and everything worked just fine.
Also the transaction you provided is a raw bitcoin transaction, not PSBT, so I am wondering how did you get it.
Can you provide full code or an example of PSBT that is signed incorrectly?

motorina0, 2022-07-30:
Thanks for looking into it. Code sample can be found [here - parseBase64Psbt](https://github.com/lnbits/hardware-wallet/blob/main/wallet/700_commands.ino#L173) and [here - sign](https://github.com/lnbits/hardware-wallet/blob/main/wallet/700_commands.ino#L197).
   - same happens when only one `sign` operation is executed

The code is running on this [device](https://www.aliexpress.com/item/33048962331.html)

The provided tx is the extracted one. Here is a new one:

Signed PSBT (8 inputs) :
` cHNidP8BAP2TAQIAAAAIDe6wQ0S6Ks/X4bZiYEOTyn6HI9x++ZWGyDxS6xACOSYAAAAAAP////+YfgDZSR1LMZxLZdnn+jx7gGNETpHoSg34Ub3jqn2XeQAAAAAA/////5h+ANlJHUsxnEtl2ef6PHuAY0ROkehKDfhRveOqfZd5AwAAAAD/////mH4A2UkdSzGcS2XZ5/o8e4BjRE6R6EoN+FG946p9l3kBAAAAAP/////eUmrMbsyJK1Kj94blkF0M6ZIJ3ia7yk2VMSH04+60WQEAAAAA/////5h+ANlJHUsxnEtl2ef6PHuAY0ROkehKDfhRveOqfZd5AgAAAAD/////mH4A2UkdSzGcS2XZ5/o8e4BjRE6R6EoN+FG946p9l3kEAAAAAP/////uG37i8OTt1oju6FxelIQNGifXdGYfdOPHs2b8tJI9BAAAAAAA/////wIw5gIAAAAAABYAFI5DLQiEzTvl+OnX+wgRu8+Dwvma8xcAAAAAAAAZdqkUjraGOza6RPwb5fjZ+/Zer235R4eIrAAAAAAAIgICTTOSdV7ar/OT4oEqXg1NmCJPVvgd+q8NdQksZkufs/lHMEQCIDqk7kpDLWJ6orSBX17s0O02SjYxVH0vS3aCNDvN4Mi5AiA7yGhmOvsBLAaKIUvXgbkY49J+5cPksxeHtjhNiBRnugEiAgMEo4ULBpdvpzh+V97HfgpWhl2A10uvnFafRsmQPFxJbEcwRAIgS5/69K1ceBRoQmNe88ynsqNPPfCiRABT8F8HIZXr0L0CIH/wceLpLqcZJ8oNQVj6dMuJ5gtDF1NfKuyfyII8hoGQASICA5LWFVduwmEOuZL0izDKk9Q8Tpi1IvTWlnRxIL5ZD7kHSDBFAiEAoZ+rh5wtlIqEg+vK2UekmXDGRPjkX0CE8lmM9CbGExYCICDO0nxxDNojKe92Wo4bzdZejmo5tAL5aafVVV5wDUiOASICApspzgwpYdDCD69U8pcAcMeOoYbKzWR8IHwlsFEIfvvrRzBEAiAA/PVQ+bUMyWKa4/29AZ+4EK62/WpXSYhWMoZkzsOsJAIgb7RcsLks2SM2Fifu0Phidzb31qbsMSwvS5tPog2/VigBIgIC8JbDKln26XJvZTeIClQHoepWck2qhs8mAbYfMvhgK2BIMEUCIQCxseDYytxLSgpXI3nmO8rLDaqxpCtBk/HlzhsbiEkbMAIgLeQ36/g8JYG3F3jiqdbLTDHniP2ycar6CAD6mlKHMeQBIgICxG46E5AezKW6YdXuo/Ww5kiTglT+O8+OoJyzgzxlgLBIMEUCIQCziBsFdOj/P8AEAzbJ5j0nN9Gc1EIzth5+r0Q2IrW5TQIgD/fGC+VLIzQTg0OYAjCW+Ybx/1YuIr0lR1pbduFDTOIBIgICjG9/Ve/UBAgXhnRbI+9rZ8QqmfmxY6rkWzhBZLhMQQRIMEUCIQD+3Zk7DiuwIY3UM0BYLFT0u6FnFv+QoRuXST4IzvMStwIgWNwoMnBDYTZVL5A6noPiNAWaSOAaccIOtzVJVDQcziYBIgID47UODj29CBSmYJkOVpu5trXCd31vptUhJoL9rWwFDFxHMEQCIEMl5Z31jdcFy0lCOlB81OthLRYZd+C2O6uhTjRfoCZPAiBNaEf4gtah/4KjxZxmljIAem6MlXht+gU+GDzyl6AqqwEAIgICTTOSdV7ar/OT4oEqXg1NmCJPVvgd+q8NdQksZkufs/lHMEQCIHiHBy15u4mA7Q3PN5kixnm+rfncEGe7IF+QAs2JR3P2AiBevOsvR01xYKO/5QmtO1IuGIyALrCwECRqzpY7rYkzHQEiAgMEo4ULBpdvpzh+V97HfgpWhl2A10uvnFafRsmQPFxJbEcwRAIgeXP6YArYmGiFnCOVKMeVcg5i+efc7Bgh7r4Ga+eXYPQCIDD9A/M+MiVQfn+pM35fUSSKiAfQwUxu1fizyQiIez3YASICA5LWFVduwmEOuZL0izDKk9Q8Tpi1IvTWlnRxIL5ZD7kHRzBEAiAmVkAusQ9M5HH69yE1/qbK/vfj+/K3otrtDNsGudLMLAIgEASscdUkZkSAhb4qyHnuZgltPrOJtbJWg/ZYc7hCVR0BIgICmynODClh0MIPr1TylwBwx46hhsrNZHwgfCWwUQh+++tIMEUCIQC7Y3tp5oXoVTOaxv/ynd6G7o/xZ07yYk8Vq3Yd8O/CrQIgeSqGYXm7OA+ViRjunIn8XMiBq41qZVfBveJhJD6+cIQBIgIC8JbDKln26XJvZTeIClQHoepWck2qhs8mAbYfMvhgK2BHMEQCIA6EPIQh1tDY9bYNNXykOQJn/jp60yeYfhTA7BDXERP5AiAGXo+2u1BfxvJJWetzHp+rn/BKBOh9nxffHrqZ49HGrQEiAgLEbjoTkB7Mpbph1e6j9bDmSJOCVP47z46gnLODPGWAsEcwRAIgCC7808t2nI6FSJ259iuCVmebBc8Fy0nwLq5Lg+Ri7N0CIAbg0ScSHoWEiLjU6ILAhd7cyvhECjH3oj+ezqe2TnepASICAoxvf1Xv1AQIF4Z0WyPva2fEKpn5sWOq5Fs4QWS4TEEESDBFAiEA4c7GzO4MPt8Bpcjruhjo5wO0qOf6XSmVM8g3Nb+S6OICIFTOjcpxLBp7sPmmNhsS9c1oxkB0usNrGp4OWBXQ7DRIASICA+O1Dg49vQgUpmCZDlabuba1wnd9b6bVISaC/a1sBQxcSDBFAiEAo1MCPRV9SZ2gGyVfoJs2RpcX56qgblEc9GLO2FFNF6kCIFd7BWHrd/bG/F+bJV4cCVvGvMmAlo01YbvJHn2SBO6bAQAiAgJNM5J1Xtqv85PigSpeDU2YIk9W+B36rw11CSxmS5+z+UcwRAIgMQ28zads0S4xloDq7QYyaEBcTSg9vSwfBT8a6M39344CIFf/JX4xBJkp6C5t8URVEt5e9+Ydi48S5mrk58pl1cHGASICAwSjhQsGl2+nOH5X3sd+ClaGXYDXS6+cVp9GyZA8XElsRzBEAiBlg0YECCuu6lxssBmgnTNGU5FDSgEBimh8at1YsZ/icgIgcK/1t7ArQAc/9Mvc7JOZf0roI7xUxeBwEB8oiL10loMBIgIDktYVV27CYQ65kvSLMMqT1DxOmLUi9NaWdHEgvlkPuQdHMEQCIExuSp4HNmD95H76i08nrwqPVyhqp61R8jDD76B9zLb2AiBdfCgC5hbblQ8I6Q1UZu6HjrN6veHvLJUSTXQDQQKw7gEiAgKbKc4MKWHQwg+vVPKXAHDHjqGGys1kfCB8JbBRCH7760cwRAIgKJvteQlBdEbusOZkF858FlvPKBWlX2lwFGVv00yvbPICIE3raajjnYhxzX0oxQ6FkyfMz62l0u0Y0C1eRpBI+2mBASICAvCWwypZ9ulyb2U3iApUB6HqVnJNqobPJgG2HzL4YCtgRzBEAiAVUXvtilIK+kmHbF7/ecRokaO/ggSTPG4Jwe2Xn6IfZgIgL9vuA10Pjod6WTmIqm7ljkqhkYiaXaYJHOsQHZtGgMYBIgICxG46E5AezKW6YdXuo/Ww5kiTglT+O8+OoJyzgzxlgLBHMEQCIFsmFhfvsjTsJsTQmyju4aPPJAVHXIhhnTOW0+p77NEYAiA/ITuQhYmfnSZIhcBAfXfL2ddQNknt6KhzvrtsY1g1qAEiAgKMb39V79QECBeGdFsj72tnxCqZ+bFjquRbOEFkuExBBEgwRQIhAJ5O127dTIZGeS6nU8nOyWAymYkda9dNPEVtTb5FqyCxAiAr1LknErcUKY3vZjkt1F25bz5g73N0E9AbYqXns5OCBAEiAgPjtQ4OPb0IFKZgmQ5Wm7m2tcJ3fW+m1SEmgv2tbAUMXEcwRAIgSpjBknFLdGRf963alhiPWBGUFM3COH0PhgZS4XyajXYCIDJae3sQPIM/meldcKS7gNmdnrfC6bv7oO7ywPKpU7nwAQAiAgJNM5J1Xtqv85PigSpeDU2YIk9W+B36rw11CSxmS5+z+UgwRQIhAMcOR+jNZphPTVWPr/dibWCEbhHiCQpv5G3fn36AKvXMAiAMRuft43e66YmRzZlrpkaWMRh1CLMmqLrhvh4fEQ4NKQEiAgMEo4ULBpdvpzh+V97HfgpWhl2A10uvnFafRsmQPFxJbEcwRAIgSCZOpcyxkpV7A0ARjqerS2JUv0RpRX26EjEa6CUhikwCIBtv2SF6KT++W2z3J/X7+nccBD1yC7NJPx7UxFthCurGASICA5LWFVduwmEOuZL0izDKk9Q8Tpi1IvTWlnRxIL5ZD7kHSDBFAiEAlyLV9tttsFmV3WQF17tS75h8WiAP1qIMRsJOoavmrzUCID/gSbfyQ/W4Rcm3Qq8yruurUVmx1MwErZAD53YQPyhgASICApspzgwpYdDCD69U8pcAcMeOoYbKzWR8IHwlsFEIfvvrSDBFAiEAk7CA8CiCLn+nktlB0Phyt0F7Uh8gLP++eEsaVyG5IbkCIFSN8EWHW1Y8ggaamSp21OYcRnZNUS37BjLbEmuzbkKnASICAvCWwypZ9ulyb2U3iApUB6HqVnJNqobPJgG2HzL4YCtgSDBFAiEApCSUcHB373U5YXVlqU/Vh4xhFfp3FfV7fe32nslWeRQCIELSQAIXwHuQbfZ3F3eo6h4WyELPAtYfj3/VYSEGgUqjASICAsRuOhOQHsylumHV7qP1sOZIk4JU/jvPjqCcs4M8ZYCwSDBFAiEAkpCAS1BpfRfVwBW3wlHV8ChE9VST6K3kFpOBPDf59joCID1vEDMo4t/jWszyXzHyiaiYkzQgW7bfkswTVDpC4kE5ASICAoxvf1Xv1AQIF4Z0WyPva2fEKpn5sWOq5Fs4QWS4TEEESDBFAiEA/+6WCElQJf8T2soSZsRtJJz3QaiIIovUw2tiJO124LICIAFk9lkjS1X4NqbnPcmvxVgFnQPl1G6s2qJY3Uc+dz0tASICA+O1 [... cut ...] mMwwXC5jBfaF6WagIgaJTw8AA5E+DNnzBddvNkm0bDum7bQbkb7AxsKP28LdIBIgICxG46E5AezKW6YdXuo/Ww5kiTglT+O8+OoJyzgzxlgLBIMEUCIQDE9IzECYDcjVH+3L6arGmEmrUyCZIpFv03HAHi+KS7hwIgL/JF2znnIQlrDR1dLbj7RH0iwzud8wHDhicx+/UFhXIBIgICjG9/Ve/UBAgXhnRbI+9rZ8QqmfmxY6rkWzhBZLhMQQRHMEQCIF/KxBHh+IgHev8V/eaIuywHST0G+lAYYBFXCcCido00AiBYJqomNyxHcGe9fw05yVK5tuJR0S/FC7iLhSdVUXsJHwEiAgPjtQ4OPb0IFKZgmQ5Wm7m2tcJ3fW+m1SEmgv2tbAUMXEgwRQIhALzWSgAdm/MNkIAgfP8HiWVJemSDVveIwoQIaE6wvJBPAiB9z+57JRLAet589eryD0DpL+yGSWRrvkCoQffvrgUDZwEAAAA= `

motorina0, 2022-07-30:
Sorry for using you for [rubber duck debugging](https://en.wikipedia.org/wiki/Rubber_duck_debugging)
I was building the PSBT incorrectly (the `bip32DerivationPath`) and this lib was matching inputs multiple times while signing.
It looks good now, thanks!

stepansnigirev, 2022-07-30:
Actually, thanks to this issue I found another bug - if you sign the transaction twice the signatures are also added twice even though they are the same. It makes PSBT invalid. Gonna fix it soon.

closed 2022-08-03 by motorina0