micro-bitcoin/uBitcoin #27: Version 0.2.0 is not available at Platformio

ktecho, 2023-05-31:
Last version available is 0.1.2 which is very old.

BitMaker-hub, 2023-05-31:
Hi Luis, did you tried adding the github repo directly to lib_deps on platform.ini?
This should use last release version

```
lib_deps  = 
	https://github.com/micro-bitcoin/uBitcoin.git
```

ktecho, 2023-06-01:
> Hi Luis, did you tried adding the github repo directly to lib_deps on platform.ini? This should use last release version
> 
> ```
> lib_deps  = 
> 	https://github.com/micro-bitcoin/uBitcoin.git
> ```

Yes, the thing is that I added the dependencies to this project for the same versions as in the Arduino library and the failed:

https://github.com/lnbits/arduino-nostr/pull/8

I wonder if  I can specify the github repo as a dependency...

blackcoffeexbt, 2023-06-08:
ty for the PR btw, as soon as uBitcoin 0.2.0 is on the platform.io registry, let's get this PR merged!

stepansnigirev, 2023-06-08:
Updated the library on platformio to v0.2.0

closed 2023-06-08 by stepansnigirev