micro-bitcoin/uBitcoin #30: uBitcoin/src/BaseClasses.cpp:3:10: fatal error: cstdlib: No such file or directory

FeatureSpitter, 2023-10-19:
I installed this library using the visual studio code Arduino extension (by Microsoft). And I am getting this error when compiling:

[image]

```
[Starting] Verifying sketch 'arduino-wallet.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
/home/nemewsys/Arduino/libraries/uBitcoin/src/BaseClasses.cpp:3:10: fatal error: cstdlib: No such file or directory
 #include <cstdlib>
          ^~~~~~~~~
compilation terminated.

Error during build: exit status 1
```

All the other libs seem to work fine, this one is the exception.

Do I need to make anything additional to have this working with VS Code?

Thanks!

FeatureSpitter, 2023-10-19:
@stepansnigirev

FeatureSpitter, 2023-10-19:
I get the same error in the IDE it seems:

[image]

I am using the latest version of the IDE, the latest version of uBitcoin and Ubuntu 22:

Linux nemewsys-82JU 6.2.0-34-generic #34-Ubuntu SMP PREEMPT_DYNAMIC Mon Sep  4 13:06:55 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

closed 2023-10-20 by FeatureSpitter

tinygiant, 2023-12-22:
How was this resolved? I'm seeing the same thing on a windows 10 pc.