micro-bitcoin/uBitcoin #33: bug: redeclaration of `Network` on esp 3.x.x

dni, 2024-07-30:
esp32 3.x.x brokeb uBitcoin

```In file included from /home/dni/Arduino/libraries/uBitcoin/src/Bitcoin.h:11,
                 from /home/dni/repos/lnbits/lnpos/lnpos/lnpos.ino:16:
/home/dni/Arduino/libraries/uBitcoin/src/Networks.h:42:3: error: 'typedef struct Network Network' redeclared as different kind of entity
   42 | } Network;
      |   ^~~~~~~
In file included from /home/dni/.arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/Network/src/Network.h:10,
                 from /home/dni/.arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/WiFi/src/WiFiGeneric.h:39,
                 from /home/dni/.arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/WiFi/src/WiFiSTA.h:29,
                 from /home/dni/.arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/WiFi/src/WiFi.h:33,
                 from /home/dni/repos/lnbits/lnpos/lnpos/lnpos.ino:1:
/home/dni/.arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/Network/src/NetworkManager.h:34:23: note: previous declaration 'NetworkManager Network'
   34 | extern NetworkManager Network;```

Uefi1, 2025-02-03:
Hello, I also encountered this problem, were you able to fix it somehow ?

oren-z0, 2025-05-31:
I'm stuck on this issue too.
I just wanted to use `NostrRelayManager.h` (from https://github.com/lnbits/arduino-nostr) with `WiFiManager.h` (from https://github.com/tzapu/WiFiManager).
Types should have a distinct prefix, or use namespace.

dni, 2025-06-10:
as a workaround we are still using esp v2 for arduino